bitluni/ESP32Lib #2: Requesting an MIT license for the I2S interface code and DMA buffer

Zvoc47, 2019-02-11:
Hello. I've been watching your videos and I've decided to buy an ESP32 and a parallel ILI9341 display since it seems that ESP32 is the only microcontroller fast enough to drive a full 320x240x16bit@60fps display. I would like to use your I2S interface code and the DMA buffer code, but my project is going to be licensed under the MIT license which is incompatible with your ShareAlike license. Could you please give me an MIT license for this piece of code? I would also be very thankful if you could also give me instructions on how to drive my ILI9341 display with that piece of code. I'm thinking of using a scanline buffer rather than a framebuffer too and that requires very precise timing to prevent screen tearing.

Thank you.
Best regards,
Zvoc

bootrino, 2019-02-11:
I'd prefer MIT license too for the whole library, and I respect also if bitluni would prefer not to.  

I do think very permissive licenses tend to result in software that is more widely used.  You'll never know how many projects it didn't get used in because the license isn't commercial use friendly.

ristomatti, 2019-02-11:
I would think there are also other commercial friendly licenses  MIT is quite permissive.

bootrino, 2019-02-11:
The other VGA lib for esp32 https://github.com/fdivitto/fabgl is GPL but I'm not very enthused about GPL in commercial products. Not that I have any commercial products planned with this library but you never know what might come to mind!

bitluni, 2019-02-11:
TFT support might come later.
The by-sa license is commercially friendly. It only enforces remixes to be open source as well and reference the original. 
I can understand that it might be an issue with the compatibility to MIT.
Still, I have put in many work hours in this project not knowing how to pay my rent in two months from now. If anyone likes to use the library for commercialization w/o publishing the source or giving me credits he can purchase an individual license so I can buy me food 😅.
Another aspect is the support. I'm currently the only one knowing every part of the code being able to fix any issues and ensure best performance. Making it too open ten million clones would appear in the arduino library manager with different faulty states of the code making the experience of the users miserable.
However, I'm still open to support individual non commercial projects for that require different licenses. Thatfore I need to know the purpose so I can write an idividual license.

Wow, where did fabgl appear from all of the sudden?

I just checked the webz regarding gpl. I will update to cc by-sa 4.0 which is compatible with gpl.

bootrino, 2019-02-11:
>> purchase an individual license so I can buy me food

In a strange sort of way I think that projects that make themselves most commercial-usage friendly can end up being the ones that can make money, not through selling commercial licences, but because being most commercially friendly makes them the market leader and dominant player - the most obvious choice for all applications.  

The financial benefit comes from being the main market player.  Consider welcoming people who come to make money from your software.  If you play it right, you will benefit from being the one true source, and all those extra commercial players using your software will only amplify the focus on you, and thus you sell more luniboxes or lunimachines or whatever you call them.

It's sort of the same topic, but regarding FabGL and whether or not you use MIT license ....... the important thing is that you are popular, people like you, you have an audience, you have a channel to promote the bitluni multimedia library.  The wider you spread that message, the better for you.  MIT license spreads it wider.

You are the central point.  Make a great BitluniMachine.  Start by making an esp32 board with integrated VGA.  You are a creative guy - make the next version including a creative, fun and funny beautiful plastic case (commercially produed, not 3d printed) - an integrated machine.  Make lots of videos about your machines and your library (and by the way, you are doing alot of development I can see right now - you should be doing alot of videos too - where are they?  don't bother with quality - keep your audience engaged and up to date about the exciting developments - daily update videos about new code features - quick and dirty video production values).

And yes FabGL ha turned up seemingly fully formed, but I suspect it's because you have published first and in the most public way.  You have the audience - they do not.  With all due respect to FabGL, grab all their best ideas and put them into bitluni .... all the tech companies have done this for years. Your system is better anyway because you require no additional components.

So I don't think much money would come from a restricted licence which requires others to buy commercial rights and therefore feed you - seems to me the most successful software these days comes from those who completely welcome others using it commercially, knowing that every commercial user cements them as the central market player.

And regarding FabGL - just do better than them and grab all their best ideas - that's the way software has always worked.

bitluni, 2019-02-11:
Thanks for all the considerations. I thought some time about it before I decided to do it like that. 
A common viewer or a maker doesn't care what kind of license it is. It's still an **open culture** license and anyone can use it even for commercial products.
I have published most of the parts under public domain before. but you had to do your homework to use it. What I'm doing here is delivering a complete package now and want to keep being part of it. 
You seem to have a strong personal opinion about this licensing stuff and I really don't. 
To be clear: **You(and anyone else) is allowed to create a project that uses the library and publish it under any license as long the library isn't cloned. The compilation happens on the user side not on yours.** If you want to distribute a finished compiled firmware/library etc that has my library in it, you need to publish your sources and attribute the library properly.
example:
You created an Arduino project that simply needs the ESP32Lib to be installed in the environment. You can publish your sources under MIT (since you don't publish parts of the lib)

Regarding the BT.. please don't give it all away now ;-)

Zvoc47, 2019-02-11:
> TFT support might come later.
> The by-sa license is commercially friendly. It only enforces remixes to be open source as well and reference the original.
> I can understand that it might be an issue [...]

I see the point. However, ShareAlike is dangerous. I remember wanting to use an amazing Super Mario platformer engine, but it was GPL because the owner used many components and only one of them was under the GPL license. The owner would gladly give me the MIT license, but the GPL cancer is there. So that way, one day when I make my game console and people make games, they'll complain to me why the license is GPL and it will turn out that I had to make it GPL because of someone else because of someone else because of someone else because someone injected the GPL cancer. As a matter of fact, GPL license is like giving the cake and eating it in front of those to who the license is given. MIT license gives credit, but so does Apache License 2.0 so you don't have to worry about that. I've seen a lot of your code, but never an I2S bus that's used for rendering an image. Why cannot I have this tiny little bit?

In the meanwhile, you've replied with this:

> Thanks for all the considerations. I thought some time about it before I decided to do it like that.
> A common viewer or a maker doesn't care what kind of license it is. It's still an **open culture** license and anyone can use it even for commercial products.
> I have published most of the parts under public domain before. but you had to do your homework to use it. What I'm doing here is delivering a complete package now and want to keep being part of it.
> You seem to have a strong personal opinion about this licensing stuff and I really don't.
> To be clear: **You(and anyone else) is allowed to create a project that uses the library and publish it under any license as long the library isn't cloned. The compilation happens on the user side not on yours.** If you want to distribute a finished compiled firmware/library etc that has my library in it, you need to publish your sources and attribute the library properly.
> example:
> You created an Arduino project that simply needs the ESP32Lib to be installed in the environment. You can publish your sources under MIT (since you don't publish parts of the lib)
> 
> Regarding the BT.. please don't give it all away now ;-)
> 
> > And yes, offer support plans at @$5,000 per month to commercial companies using LuniBits, but no charge to use it cause its MIT.
> 
> How does that make a difference to a closed commercial license?

So if I take out the I2S+DMA code only and use it in my source with the MIT license, is that okay?

And to clarify, I would like to make a game console that has this LCD speed issue solved once and for all and where people can make their own games for it by using MicroPython on the go or some emulator for a fantasy console like Pico-8. I'd write the graphics engine as I've already begun and license that under the MIT license so that people are free to make their own games without the fear of copyright issues that the GPL/ShareAlike and other viral copyleft cancer licenses impose. As for the games' licenses, everyone should be allowed to use any license they want. For example, I'd license my games under a special fandom-friendly family-friendly license that allows fandom works which are PG-13 rated in order to prevent disgusting derivative works that are responsible for all the cringes in online fandoms. A ShareAlike would just make it easier to propagate inappropriate content and that would not be good. Alternatively, I could use something other than the MIT license where everyone has some common-sense fandom rights which I'll specify according to what the thriving fandoms have always been using and where all commercial works' revenues get split across all contributors equitably so that everyone gets a bit of glory without having to trade their passion for it and so that my friends can finally tell their parents that this fandom stuff isn't just for fun, but for real money so they can move out to be independent and to meet with their friends in real life to then turbo boost each other's creativity to make amazing content!

You were not expecting that, were you? Well, now you know what my motives are. May I please get some help for my parallel ILI9341 display now?

ristomatti, 2019-02-12:
A lot of demands for creating something that's fun for people to play around and hack with for free. Isn't that enough?

Zvoc47, 2019-02-13:
Free and non-commercial is one thing. But when this "freedom" becomes copyleft, that's when problems begin.

bitluni, 2019-02-18:
Sorry guys. I just need a head start on this here. Once I happy with everything I'll change the license. I will only give proprietary licenses to specific projects now.