bitluni/ESP32Lib #3: Printing "special" methods get strange results

Gun-neR, 2019-02-19:
I love this library!!!... Thank you!!

It is much easier for me to experiment with.   So of course I used it to continue an experiment I have of dumping Serial debug text to various displays when using the Blynk IoT App

Anyhow... the story is here - 

https://community.blynk.cc/t/my-attempts-at-transferring-blynk-print-to-various-displays/21649/37

But the issue/question I have is how to `vga.print(char(<ASCII value>))` and actually get the character and not the decimal value?

I also noticed some strange buffer dumps or something when printing things like this `vga.print(uptime + ", ");` as in my example in the link.

Yes, I know I am using this library in a strange way... but hey it sorta worked :)

[image]

Gun-neR, 2019-02-19:
Hmmm... I found some printing Text hints in your VGA2DFeatures demo... I shall experiment further...

Gun-neR, 2019-02-21:
OK, I think my "issue" is not as severe as I thought.  Basically I was assuming that `vga.print()` would act exactly as `Serial.print()`... and maybe it will in future iterations?

Meanwhile I dug into the examples and some Google-Fu later, I managed to utilise this library with character arrays to function perfectly for my experimental project.  

So this can be more of a _"Look what I did with this VGA library"_ thing than an issue.  Feel free to close it as needed.

Thank you Bitluni

[image]

closed 2019-02-21 by Gun-neR

bitluni, 2019-02-21:
It's awsome! 
I'll try to extend the functionality of the print functions.. how could I miss implementing char!?