bitluni/ESP32Lib #4: VGAWiFiTextTerminal_demo not conect

LilyGO, 2019-02-20:
I tried a lot of routines and the display works fine. But only this program, the IP address generated by the phone connection: 192.168.4.1 cannot be connected. The display has been shown to connect to WiFi and generate an IP address, and the phone is connected to the same WiFi. However, the page cannot be opened.
I hope I can get your help, thank you.

Gun-neR, 2019-02-21:
**It should be as simple as loading the sketch, looking for VGA in your phones WiFi AP options, 
Connect to this AP, then using a browser on your phone, connect to 192.168.4.1.
Then type away...**
[image]

[image]

[image]
[image]

bitluni, 2019-02-21:
@LilyGO did you tr connecting to your wifi instead?

thx @Gun-neR 
that's what I expect

closed 2019-02-28 by LilyGO