bitluni/ESP32Lib #5: You have to connect the VGA cable after the ESP turns on

ShonP40, 2019-02-20:
You have to connect the VGA cable after the ESP turns on.
If I am connecting the VGA cable and then powering on the ESP I will not get signal but if I let the ESP boot and then connect the VGA cable I will get a signal.

Gun-neR, 2019-02-21:
I suspect much depends on the monitor you use.   Mine has an automatic adjustment based on the signal and the displayed content.  One way I found to assure my monitor locks onto the signal at ESP boot is to set the background color to something that is not black :)

```
  //clearing with coloured background
  vga.clear(vga.RGB(0xff0000));  // BBGGRR
```

ShonP40, 2019-02-21:
There is a blue LED on the ESP that lets you know when the code is running.
But when I power on the ESP after it’s already plugged into the monitor the blue LED doesn’t turn on.

Gun-neR, 2019-02-21:
I can't speak about the LED... mine seems to be always on.  When I started playing with this library, I also thought it wasn't working.  But as stated, it was the monitor not catching the signal in time, not the ESP32 or the library.

bitluni, 2019-02-21:
the problem is gpio 0
all my monitors don't mess with it but some people were reporting having problems. 
change from gpio 0 for hsync to 32 or 33

ShonP40, 2019-02-22:
Thanks. I will try it later today.

ShonP40, 2019-02-23:
Changing hsync from gpio 0 to gpio 32 fixed the problem.

Thanks.

closed 2019-02-23 by ShonP40