bitluni/ESP32Lib #6: How to switch to VGA font?

bootrino, 2019-02-21:
The VGA font here looks much more readable than the Commodore font currently used.

https://www.youtube.com/watch?v=n5c27-y5tm4

What would I need to do to display with a VGA font set?

thanks

bitluni, 2019-02-21:
the font needs to be converted to an array like the current ones
did you try the 8x8 font yet? it's the gameboy font. the conversion can be done using an old converter... not sure which one.

did you get fabgl running? it doesn't work with my esps. I was wondering how Fabrizio managed to get sufficient clock rate to get 640 horizontal.

do you have a source for a good font I should take?

bootrino, 2019-02-21:
I think it's code page 437

https://en.wikipedia.org/wiki/Code_page_437

The page above has a bitmap: https://upload.wikimedia.org/wikipedia/commons/f/f8/Codepage-437.png

There's some interesting info on VGA fonts here: https://int10h.org/oldschool-pc-fonts/fontlist/

No I have not tried to implement Fabrizio's VGA - I've been fiddling with the deep pain of bluetooth on the esp32 - as I mentioned to you via Twitter, consider going directly to btstack if you're ever doing any bluetooth audio/keyboard/mouse/gamepad - the esp32 bluetooth is just a memory hog.

bootrino, 2019-02-24:
>>did you get fabgl running?

This guy seems to have got it running.... just right click and hit translate in Chrome

https://silentworlds.info/2019/02/24/post-721/

bootrino, 2019-02-24:
I see you put fonts in - very nice.

closed 2019-03-08 by bootrino