bitluni/ESP32Lib #9: vga connector

techno94, 2019-03-02:
hello,
just for information
there is some VGA connecteur maybe easier and quicker 
[image]
[image]
i don't know where to post his information so i make an issue

bitluni, 2019-03-03:
thanks for sharing.. here it's ok. I have ordered B type. do you have a link to to the upper one?

techno94, 2019-03-03:
here it is : 
https://fr.aliexpress.com/item/Connector-DB15-D-SUB-VGA-male-nut-Terminal-breakout-Plastic-Cover-3-2-Data-Cable/32636892170.html?spm=a2g0w.search0104.3.117.e68d73a4sVFDyJ&ws_ab_test=searchweb0_0,searchweb201602_6_10065_10068_319_10892_317_10696_10084_453_454_10083_10618_10304_10307_10820_10821_537_10302_536_10902_10843_10059_10884_10887_321_322_10103,searchweb201603_56,ppcSwitch_0&algo_expid=cdbb6c89-abb9-4ca8-8573-6844fda1676f-14&algo_pvid=cdbb6c89-abb9-4ca8-8573-6844fda1676f

closed 2019-03-05 by bitluni