bitluni/ESP32Lib #10: esp32 VGA visual programming

techno94, 2019-03-03:
Hello,
i've made block for ardublock for programming the vga 
i will publish when it's debug (it s in french and english, the screen is in french)

[image]

and i ve made an app to have direct link in arduino to the sprite editor

[image]
unzip in arduino/tools
[SpriteEditor.zip](https://github.com/bitluni/ESP32Lib/files/2923716/SpriteEditor.zip)

bitluni, 2019-03-05:
wow, that's awesome! I never used ardublock before but I will try it !

techno94, 2019-03-05:
here you can download an alpha realize with in the last dongel   "display" in the bottom the VGA block of you re librarie.
you just have to download and unzip
http://duinoedu.com/dl/logiciels/arduino/arduino_augmente/version_duinoedu/VERSIONS_ALPHA/
the ALPHA_0.70.01/    867MO  for programming arduino/ esp8266/esp32