bitluni/ESP32Lib #12: Idea: keyboard/swipe input from iPhone via QR code

bootrino, 2019-03-08:
If you display a QR code on the screen, simply pointing an iPhone camera at it will scan the QR code - no app needed.

If the qr code contains a URL then it will ask the iPhone user and open the URL.

You could point that URL to a link local address https://en.wikipedia.org/wiki/Link-local_address like http://bitluni.local

The link local named address (http://esp32fs.local) technique is seen here: https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/examples/FSBrowser/FSBrowser.ino

You can then serve a page from the esp32 to the iphone and use websockets to send keystrokes from the iPhone to the esp32.

If you got really fancy you could actually send touch gestures like swipes from the phone to the esp32 to give the bitluni graphics a touch/swipe interface.

techno94, 2019-03-08:
a qrcode is a sprite or an image. 
so with the utilitie, you could display an image of a qrcode.
https://github.com/bitluni/ESP32Lib/blob/master/Utilities/SpriteEditor.html

after there is this librarie : 
https://github.com/anunpanya/ESP8266_QRcode

who permit to display a qrcode on an oled screen with just this instruction : 
qrcode.create("Hello world.");
with this instruction, when you scan the qrcode it return : Hello world.
so it's really easy to display.
anyway, to do this, there is developpement.

bootrino, 2019-03-09:
@techno94  there's no need to generate a QR code - you just use the one above which is about 395 bytes.