bitluni/ESP32Lib #19: Higher resolutions than 200x150 or 320x240?

dl9sec, 2019-05-14:
Hi,
I am using an AZ-Delivery ESP32 DevkitC (ESP-WROOM-32) and tried out the 14Bit VGA.
The examples VGADemo14Bit, VGASprites and VGA3DEngine are working well so far, but when i try to set a higher resolution, the ESP32 runs into a boot loop caused by an unhandeled exception

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d12c7  PS      : 0x00060f30  A0      : 0x800d1328  A1      : 0x3ffb1e60  
A2      : 0x00000258  A3      : 0x00000640  A4      : 0x00000000  A5      : 0x3ffb8484  
A6      : 0x000000ae  A7      : 0x3ffb81cc  A8      : 0x00000000  A9      : 0x3ffb1e40  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000003  
A14     : 0x00000003  A15     : 0x00060823  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d12c7:0x3ffb1e60 0x400d1325:0x3ffb1e90 0x400d1335:0x3ffb1eb0 0x400eb7dd:0x3ffb1ed0 0x400eb61b:0x3ffb1ef0 0x400eb5fe:0x3ffb1f10 0x400d23bc:0x3ffb1f30 0x400d1385:0x3ffb1f50 0x400d2d2b:0x3ffb1fb0 0x400874a9:0x3ffb1fd0
```
What do i have to do to get higher resolutions? My intention was to use 800x600 for a project, at least 640x480, but currently i don't know how to get those modes work.
Any ideas/hints/solutions?

Regards, Thorsten

bitluni, 2019-05-21:
the esp32 has not enough memory for 14bit in higher resolutions.
you need to use 3bit mode or wait for the 6bit mode coming soon

dl9sec, 2019-05-21:
Ah ok, thanks for clarifying. I wasn't aware of that.
Maybe a little matrix with max resolution over bits could be helpful and avoid misunderstandings...

Regards, Thorsten

bitluni, 2019-05-21:
I'll improve the error checking to try to show a meaningful output
it really depends on your configuration and if you are using wifi or not

dl9sec, 2019-05-21:
Will the hardware architecture for the 6 bit mode be the same R2R network scheme as for the 14 bit mode?

bitluni, 2019-05-21:
you need only the upper two bits for each color
but the 14bit versions are compatible

mattsaintdev, 2019-05-21:
Try fabgl library as well.Different approach to I2S buss but supports 64 colour or 8 color modes. 14 bit for bitluni though.

Matt

dl9sec, 2019-05-22:
Already done:
https://github.com/dl9sec/ArduinoP13/tree/master/examples/PredictISSVGAsimple
;-)

closed 2019-05-22 by dl9sec