bitluni/ESP32Lib #20: question: Is is possible to use other pins ?

YordanYanakiev, 2019-05-22:
Hello.
1st of all : AWESOME WORK !!!
I would like to ask about this schematic: https://raw.githubusercontent.com/bitluni/ESP32Lib/master/Documentation/schematic3bit.png

- Is it possible to use other pins for R G B ?
( like for instance pin 25 for G, pin 32 for B and 33 for R ? )

bitluni, 2019-05-22:
sure, you only need to change them in the code as well
the pins are passed to the init method

YordanYanakiev, 2019-05-22:
Oww ! nice ! Thank You !

YordanYanakiev, 2019-05-22:
[image]
This is the circuit i've made. is there something additional which should be made for 3 bit mode ?

YordanYanakiev, 2019-05-22:
[image]
Here the circuit i've made. Is there something additional which should be done, or this is sufficient for 3 bit mode ?

YordanYanakiev, 2019-05-22:
[image]

[image]

Here the latest correction i've made in order to align as much as possible the things.
Is this looking OK, or there could be a problem ?

bitluni, 2019-05-22:
I don't see any obvious problems. you might want to provide 5v to the vga
connector as well. I'm doing this in the newer versions

dl9sec, 2019-05-22:
I would suggest to connect GND to the VGA connector pins 5,6,7,8 and 10 and put 5V to pin 9...

YordanYanakiev, 2019-05-22:
O! I've miss this probably somewhere ! Thank You very much ! Doing it.

YordanYanakiev, 2019-05-22:
[image]
[image]

This ?