bitluni/ESP32Lib #21: Halting after vga.init

hackbaellchen, 2019-05-24:
I used your example code VGAHelloWorld, added some Serial messages and tried it. They were printed until vga.init(...); the next messages are not shown and further code is not executed either.
I also tried using other pins.
I'm using the Wemos ESP32 (small blue one), that you also showed in your video.

Thank you for helping!

YordanYanakiev, 2019-06-07:
what is the resolution you are trying ?

hackbaellchen, 2019-06-07:
I'm trying 320x420. Initializing VGA is printed but nothing more.
`#include <ESP32Lib.h>
#include <Ressources/Font6x8.h>

const int redPin = 23;
const int greenPin = 18;
const int bluePin = 27;
const int hsyncPin = 33;
const int vsyncPin = 5;

VGA3Bit vga;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting");
  Serial.println("Initializing VGA");
	vga.init(vga.MODE320x240, redPin, greenPin, bluePin, hsyncPin, vsyncPin);
  Serial.println("Initializing Font");
	vga.setFont(Font6x8);
  Serial.println("Printing Text");
	vga.println("Hello World!");
  Serial.println("Done");
}

void loop() {}`

expectproblems, 2019-09-27:
I've had precisely the same problem. I am quite certain the wiring works as I've used the same pins (albeit with resisters) with FabGL lib and VGA works fine.

bitluni, 2019-09-29:
That seems serious. Did you try other examples from the library?
I could also imagine to be a power issue. Please try another usb cable
Do you have another esp32 at hands to try?

vmarguerie, 2020-09-14:
I had the same problem and solved it by adding those 2 lines in the VGAHelloWorld sketch :

#include <WiFi.h>
void setup()
{
WiFi.begin();
....
}

If someone has an explanation!

whatislife85, 2022-06-22:
I had this same problem and solved it by using the VGAWiFiTextTerminal example instead of the hello world one