bitluni/ESP32Lib #22: 1-bit implementation

jld3103, 2019-05-25:
Some projects only need black and white displays or only have such. It would be nice to have one.

Martin-Laclaustra, 2020-04-02:
@bitluni 
I implemented the 1-bit buffer VGA.
The output still needs at least 3 pins (signal, hsync, vsync), so it is implemented with 5 pins output, the same as the 3-bit driver (you can choose to use just 3, but with 5 you have the option to draw using 2 constant colors).
It would make no sense to create a high performance 1-bit driver because it would use the same amount of memory as the 3-bit one, for high performance choice, just use that one. However, for applications with constrained memory requirements, the interrupt based 1-bit driver used 8 times less memory.
This allows experimenting with higher resolutions.
I tested MODE400x600 = Mode(12, 36, 64, 400, 1, 2, 22, 600, 1, 18000000, 0, 0).
For higher resolutions, each line is not rendered fast enough. I suspect that `interruptPixelLine` function could be optimized for speed.
It is implemented in #41, which it is based on #38.

Martin-Laclaustra, 2020-04-03:
I improved the rendering algorithm and now it is faster.
It works for resolutions:
MODE640x480
MODE720x400
and
MODE512x768 = Mode(12, 68, 80, 512, 3, 6, 29, 768, 1, 32500000, 1, 1)

Martin-Laclaustra, 2020-12-21:
This is included in the "development" branch here and eventually it will be merged into "master".

closed 2020-12-21 by Martin-Laclaustra