bitluni/ESP32Lib #24: Use the library inside a thread

YordanYanakiev, 2019-05-31:
Hello. 
I am trying the standard way of working with threads, but seems like I cant make it hapen.
Is there a way to make it work on a thread ?
```
void setup()
{
	xTaskCreatePinnedToCore(
		Task1code,   /* Task function. */
		"Task1",     /* name of task. */
		286720,       /* Stack size of task */
		NULL,        /* parameter of the task */
		1,           /* priority of the task */
		&Task1,      /* Task handle to keep track of created task */
		0 );          /* pin task to core 0 */
	delay( 500 );
}

void Task1code( void * pvParameters )
{
	
	//Serial.println( xPortGetCoreID() );
	vga = new VGA3BitI();
	vga->setFrameBufferCount( 2 );
	//initializing the graphics mode
	vga->init( vga->MODE640x400, redPin, greenPin, bluePin, hsyncPin, vsyncPin );
	//setting the font
	vga->setFont( Font6x8 );
	//clearing with white background

	vga->clear( vga->RGB( 0xffffff ) );
	//text position
	vga->setCursor( 10, 10 );
	//black text color no background color
	vga->setTextColor( vga->RGB( 0 ) );
	//show the remaining memory
	vga->print( "free memory: " );
	vga->print( ( int )heap_caps_get_free_size( MALLOC_CAP_DEFAULT ) );
	//draw the logo
	//bitluni( 150, 60, 20 );
	delay( 200 );

	for( ;;)
	{
		vga->clear( 0 );
		drawMenu();
		vga->show();
	}
}
```

YordanYanakiev, 2019-06-02:
seems like in the moment this line occur :
**vga.init( vga.MODE640x400, redPin, greenPin, bluePin, hsyncPin, vsyncPin );**
the program crash. Dissregarding if it is in thread or in the setup - this line crashing the thing if there is **xTaskCreatePinnedToCore** or **xTaskCreate**

bitluni, 2019-06-05:
It's most probably a memory issue. You could try a custom resolution making
it 640x300 to test it

YordanYanakiev, 2019-06-05:
You are absolutely correct ! It didn't pass my mind at all that this could be the cause.
This is the base amount of memory i have initially with 1 thread 289.060 bytes which I have separated 1024 bytes, after I have initialized the VGA I am leaving with 8924 bytes. Therefore this mode requires about 280.000 bytes and I were giving it never the correct amount.