bitluni/ESP32Lib #25: Problems in 6 bit mode

rampa069, 2019-06-03:
Hi.
  
 I am using esp32lib in a ZX spectrum emulator (https://github.com/rampa069/ZX-ESPectrum) and it works fine in 3 and 14 bits. but when in 6 bits (dont need the 14 bits and RAM memory is a needed resource) seems that the osd is not working at all. The emulation only uses init,clear and dotfast, the osd uses rect, and print.  Are all the primitives working in 6bit?

 Thanks in advance.

rampa069, 2019-06-03:
tested one by one. vga.line is the culprit. commenting this line, all works.  This is working fine in 3bits and 14bits

bitluni, 2019-06-05:
interesting. line only uses dot itself. It's either a bigger problem or
something else

rampa069, 2019-06-06:
May be it is calculating bad coordinates? I remember to see something similar when plotting outside the buffer.

rampa069, 2019-06-12:
was my fault. had an extern to 14bit mode in a header file and this was the problem :-( thanks and sorry....

closed 2019-06-12 by rampa069