bitluni/ESP32Lib #27: [Question] ESP32-WROVER PSRAM

YordanYanakiev, 2019-06-20:
Since the VGA is using practically the whole amount of ram on a high resolutions I wonder if the lib can utilize the WROVER additional PSRAM which it have, instead of using the normal RAM?

DaniloSakovic, 2021-04-18:
I tried, without success on esp32 cam with 4mb /8mb psram, getting corrupt psram after few seconds . I assume it's interrupts... Or cache bug... Or both

Martin-Laclaustra, 2021-04-21:
Are you using "development" branch?
Try with that branch with a mode without capital "I" in the name. Those do not use interrupts at all.
Please report back.
I think exploring the use of psram is interesting.
Even for the interrupt modes (if you want much higher resolutions), an intermediate "buffer" could be created and refill outside of the ISR function (from the main loop). (if you can cope with occasional "glitches" when the main loop is late for that refill.