bitluni/ESP32Lib #28: Rotate screen - flip

RegyRegy, 2019-07-22:
Is it possible to rotate screen by 90 deg ? to left or right?

RegyRegy, 2019-07-24:
My quick work around to flip text to the left:
`	virtual void drawChar(int x, int y, int ch)
	{
		if (!font)
			return;
		if (!font->valid(ch))
			return;
		const unsigned char *pix = &font->pixels[font->charWidth * font->charHeight * (ch - font->firstChar)];
		 for (int px = 0; px < font->charHeight; px++)
			for (int py = font->charWidth; py > 0; py--)
				if (*(pix++))
					dotMix(px + x, py + y, frontColor);
				else
					dotMix(px + x, py + y, backColor); 
		}

	void print(const char ch)
	{
		if (!font)
			return;
		if (font->valid(ch))
			drawChar(cursorY,280-cursorX , ch);
		else
			drawChar(cursorX, cursorY, ' ');		
		cursorX += font->charWidth;
		if (cursorX + font->charWidth > xres)
		{
			cursorX = cursorBaseX;
			cursorY += font->charHeight;
			if(autoScroll && cursorY + font->charHeight > yres)
				scroll(cursorY + font->charHeight - yres, backColor);
		}
	}`

AsifKhan991, 2022-08-07:
@RegyRegy I needed similar functionalities. So modified the GFXwrapper library to enable display rotation and flip. Please check out [this](https://github.com/AsifKhan991/Customized-GFXwrapper-library-for-Bitluni-ESP32) repository of mine!