bitluni/ESP32Lib #30: [Question] font size

makmar, 2019-08-13:
Great project.

I order the board and I cant wait to start playing with it. 
in my project, I will need very big fonts (5 numbers will fill all the screen).
I have to create a new file in the ressources but for the moment i don't know how.  
any suggestion will be appreciated. 

thank you

bitluni, 2019-08-14:
Did you try the adafruit fonts? with .setTextSize(x);  ?
you can also use a custom resolution to simply make the pixels huge

makmar, 2019-08-29:
I received the board and works great with the examples. 
I have create a custom font file witch I store in Ressources folder with the other fonts.
I run the "hello world" example  and i #include <Ressources/newfont.h>,  vga.setFont(newfont);
when i verify the code i receive the message "newfont is not declared in the scope"
any idea what else I suld do. 

Thanks

MaxBursell, 2024-08-09:
How did you change the font size? .setTextSize(x) doen't seems to work

makmar, 2024-08-09:
If I remember,I found a program and I created my own big font set.