bitluni/ESP32Lib #31: vga.init conflict with SPIFFS?

nealcrook, 2019-09-03:
I am trying to use a SPIFFS filesystem in conjunction with your vga library. If I do all of my SPIFFS calls before doing vga.init() everything seems to work fine. However, if I try to do any SPIFFS operations after vga.init() I get crashes and core dumps

I am using pins 21, 22, 18, 19, 4, 5, 23, 15 for VGA

For example, the following code fails, but if I mode the vga.init call as indicated, it all works fine

if(!SPIFFS.begin(true)){
        Serial.println("An Error has occurred while mounting SPIFFS");
        return;
    }

Serial.println("About to call vga.init");
vga.init(vga.MODE400x300, redPin, greenPin, bluePin, hsyncPin, vsyncPin);
    
unsigned char * ram = nas.ram;
File file = SPIFFS.open("/lollipop.bin");
long len = file.size();
file.read(ram + 0x1000, len);
file.close();  

// <--  move vga.init call to here and it works fine -->
    
by "fails" I mean the arduino serial console reports:

About to call vga.init
I2S TX
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400f53d0: bad00bad bad00bad bad00bad

I don't want to be rash and call this a "bug" but I'd be interested in any theories of why your vga library would interact badly with SPIFFS

thanks.

Neal.

bitluni, 2019-09-04:
Interesting effect. I can only imagine the data bus is too busy to handle
additional load. does it happen an any resolution?
Are you using the interrupt based implementation?

nealcrook, 2019-09-04:
Thanks for your reply. Running out of bandwidth is an interesting idea. I am using a "no frame buffer" implementation based on your "VGANoFrameBuffer" but using VGA3BitI rather than VGA14BitI. I am running using MODE400x300. I am actually implementing a "character mapped" rather than "bit-mapped" display, so InterruptPixelLine creates pixel bit data by looking up the character in a "character generator" array. Overall, I don't think it should be more memory-hungry, but I will try running with a clean version of your VGANoFrameBuffer", just adding SPIFFS calls, and report back.

bitluni, 2019-09-05:
Ok, that's the interrupt based version of the driver. That one is notorious
for not working with other stuff since it needs a lot of cpu power.
Not sure any more if it's running on the second core. If so it might cause
problems with the buggy cache of the esp32. WiFi and stuff doesn't work
either with the interrupt version.
I'd use a single frame buffer with VGA3Bit and do the conversion of you
character field in the main loop. won't flicker either. It might be even a
better practice since you probably are not updating the
image 60 times/s. The only reason to use the I version would be if you
REALLY need all the memory. since 400x300 3bit needs like 120kb you are
still left with about 80kb.

nealcrook, 2019-09-05:
Yes, reading more it seems that interrupts and SPIFFS don't work well together unless you pay special attention:
https://www.esp32.com/viewtopic.php?t=8597
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/fatal-errors.html#cache-disabled-but-cached-memory-region-accessed
I can probably get by with 80kb so I'll have a play with the non-interrupt version. Thanks
Neal.

PR #37 (2020-05-17): Fix for incompatibility between interrupt VGA modes and SPIFFS/WIFI
Fixes interrupt clearance.
Interrupt() call to child replaced by pointer to a static function with instance as argument.
IRAM flag in routines.
Also fixes an error in rendering in VGA6BitI.
Fixes #31
closed 2020-05-17 by bitluni