bitluni/ESP32Lib #33: Can't compile any example sketch

CommandActor, 2019-10-31:
I'm trying to compile the VGAHelloWorld example, but every time I try to upload it, the Arduino IDE just fails.
I'm using Arduino IDE 1.8.10

`In file included from c:\users\varga\documents\arduino\libraries\bitluni_esp32lib\src\i2s\DMABufferDescriptor.h:13:0,
from c:\users\varga\documents\arduino\libraries\bitluni_esp32lib\src\i2s\i2s.h:23,
from C:\Users\varga\Documents\Arduino\libraries\bitluni_ESP32Lib\src/VGA/VGA.h:14,
from C:\Users\varga\Documents\Arduino\libraries\bitluni_ESP32Lib\src/VGA/VGA14BitI.h:13,
from C:\Users\varga\Documents\Arduino\libraries\bitluni_ESP32Lib\src/ESP32Lib.h:3,
from C:\Users\varga\Documents\Arduino\libraries\bitluni_ESP32Lib\examples\VGAHelloWorld\VGAHelloWorld.ino:6:

c:\users\varga\documents\arduino\libraries\bitluni_esp32lib\src\i2s\DMABufferDescriptor.h: In static member function static void** DMABufferDescriptor::allocateDMABufferArray(int, int, bool, long unsigned int):

c:\users\varga\documents\arduino\libraries\bitluni_esp32lib\src\tools\log.h:25:59: error: exception handling disabled, use -fexceptions to enable

#define ERROR(a) {Serial.println((a)); delay(3000); throw 0;};

c:\users\varga\documents\arduino\libraries\bitluni_esp32lib\src\i2s\DMABufferDescriptor.h:34:4: note: in expansion of macro ERROR

ERROR(Not enough DMA memory);

Multiple libraries were found for ESP32Lib.h
 Used: C:\Users\varga\Documents\Arduino\libraries\bitluni_ESP32Lib
exit status 1
Error compiling for board ESP32 Dev Module.`

bitluni, 2019-11-01:
Ooh.. i'm confused about it saying multiple libraries found..
How did you install it?
Can you please try uninstalling it from the library manager and installing
it again. If it doesn't work please try the version older

CommandActor, 2019-11-01:
I installed the library using the Library Manager.
I tried uninstalling it and reinstalling it again. I also tried older versions (v0.2.2, v0.2.0, v0.0.1). All versions failed to compile. I also tried those versions in Arduino IDE 1.8.5, but it also failed.
When I try to compile version v0.2.2, the error is the same, but if I try version v0.2.0 or v0.0.1, the error is different.

bitluni, 2019-11-01:
I'll check the latest esp32 board library version. I have worked with 1.0.1
until now

CommandActor, 2019-11-01:
I decided to uninstall the ESP32 board library from IDE and reinstall the latest version. It works now! Maybe I was using a very old version. I haven't updated it since I installed it.

closed 2019-11-01 by CommandActor

bitluni, 2019-11-01:
ok, thanks for closing. my line to china was really slow today. just
managed to finish the boards update. Have fun! :-D