bitluni/ESP32Lib #42: [Question] How to connect a composite video connector in parallel output mode ?

erenard, 2020-04-19:
Hi, it seems you most recently committed [some composite stuff](https://github.com/bitluni/ESP32Lib/commit/222d95318fc5c61033dbdbd92c56f8a7d3a60129) but when I look at `Composite.cpp` you still use the [parallel output mode](https://github.com/bitluni/ESP32Lib/blob/master/src/Composite/Composite.cpp#L51).

How to connect all the pins used in parallel mode to a composite video connector ?

bitluni, 2020-04-20:
That commit was experimental, I tried to get it running with an external
resistor ladder as DAC. Similar to my VGA stuff

bitluni, 2020-04-20:
The color refused to work though...

erenard, 2020-04-21:
I'm trying to make your other repo [ESP32CompositeVideo](https://github.com/bitluni/ESP32CompositeVideo)'s code work with a composite screen I recently bought (it works with an USB video aquiring dongle) but no luck so far. That's why I want to give this lib a try, especially with all the video modes you add.
So really, even if it's only in grayscale, I would still like to try the resistor ladder.

Martin-Laclaustra, 2020-04-22:
I just finished some code to use this library with the DAC (only one connection, from pin 25 or 26). Grayscale. PAL or NTSC, fake progressive low res or interlaced high res.
I will be preparing a pull request in a couple of days if you can wait.
Also, until bitluni can review, discuss, and approve my code, I will have a branch in my fork with the code ready for download and install.

Martin-Laclaustra, 2020-04-26:
@erenard 
I pushed my changes. I created pull-request #47 . Until bitluni reviews and comments on them you can download the branch from my repository:
`git clone -b CompositeDACandLadder https://github.com/Martin-Laclaustra/martin-laclaustra_ESP32Lib.git
`
I would be glad to hear your feedback.

Martin-Laclaustra, 2020-12-21:
This is included in the "development" branch here and eventually it will be merged into "master".

closed 2020-12-21 by Martin-Laclaustra