bitluni/ESP32Lib #51: font creation

enemykrs, 2020-06-29:
hi, is there a manual or tool for creating a font?
I need to add the Cyrillic alphabet, but I can’t understand how

Martin-Laclaustra, 2020-07-02:
I am writing a converter for psf files.
Until I release it and I make a pull request... are these files useful?
[cp866.zip](https://github.com/bitluni/ESP32Lib/files/4866847/cp866.zip)
Can you please confirm that they work?

enemykrs, 2020-07-03:
thanks, but these files do not have the characters I need
I’ve already figured out how to draw a font and added the letters I need
but, if I understood correctly, the width of the letter is fixed and, for example, the signs "." and "W" have the same width, which leads to uneven gaps between characters

Martin-Laclaustra, 2020-07-03:
Is there any official codepage that contains the characters you need? 855?
Any one in: "/usr/share/consolefonts"?
All the fonts in this engine are fixed-width. For other needs use fonts through the Adafruit engine instead.
See the "GFXWrapper.ino" example.

enemykrs, 2020-07-03:
codepage KOI8-R

the example "GFXWrapper.ino" ended with a compilation error
I did not understand the reason
`In file included from C:\Users\Admin\Documents\Arduino\libraries\bitluni_ESP32Lib\examples\GFXWrapper\GFXWrapper.ino:7:0:

C:\Users\Admin\Documents\Arduino\libraries\bitluni_ESP32Lib\src/GfxWrapper.h: In instantiation of 'GfxWrapper<Base>::GfxWrapper(Base&, int, int) [with Base = VGA6Bit]':

C:\Users\Admin\Documents\Arduino\libraries\bitluni_ESP32Lib\examples\GFXWrapper\GFXWrapper.ino:13:38:   required from here

C:\Users\Admin\Documents\Arduino\libraries\bitluni_ESP32Lib\src/GfxWrapper.h:19:8: error: 'GfxWrapper<VGA6Bit>::base' will be initialized after [-Werror=reorder]

  Base &base;

        ^

C:\Users\Admin\Documents\Arduino\libraries\bitluni_ESP32Lib\src/GfxWrapper.h:23:26: error:   base 'Adafruit_GFX' [-Werror=reorder]

   Adafruit_GFX(xres, yres)

                          ^

C:\Users\Admin\Documents\Arduino\libraries\bitluni_ESP32Lib\src/GfxWrapper.h:21:2: error:   when initialized here [-Werror=reorder]

  GfxWrapper(Base &vga, const int xres, const int yres)

  ^

cc1plus.exe: some warnings being treated as errors`

Martin-Laclaustra, 2020-07-03:
> the example "GFXWrapper.ino" ended with a compilation error

Try to move lines 19 and 20 in GfxWrapper.h
https://github.com/bitluni/ESP32Lib/blob/8f4b1bc9c331546c57c1ee837ac37303b8d8739c/src/GfxWrapper.h#L19

to line 26.

Please report if this change succeeds because we should write a pull request in that case.

Martin-Laclaustra, 2020-07-03:
> codepage KOI8-R

Please tell me if this font renders correctly, and if it is the one you needed.
[koi8_r_8x16.h.zip](https://github.com/bitluni/ESP32Lib/files/4871611/koi8_r_8x16.h.zip)

Martin-Laclaustra, 2020-07-04:
> the example "GFXWrapper.ino" ended with a compilation error

After some thought, this may be the solution:
Change lines 22 and 23 in GfxWrapper.h
https://github.com/bitluni/ESP32Lib/blob/8f4b1bc9c331546c57c1ee837ac37303b8d8739c/src/GfxWrapper.h#L22
https://github.com/bitluni/ESP32Lib/blob/8f4b1bc9c331546c57c1ee837ac37303b8d8739c/src/GfxWrapper.h#L23

to this:

```
		:Adafruit_GFX(xres, yres),
		base(vga)

```
Please report if this change succeeds because we should write a pull request in that case.

Martin-Laclaustra, 2020-07-04:
> codepage KOI8-R

Now the source files for these I believe had the correct character maps:
[koi8-r-8x8-8x14-8x16-fnt.zip](https://github.com/bitluni/ESP32Lib/files/4872161/koi8-r-8x8-8x14-8x16-fnt.zip)

enemykrs, 2020-07-07:
fonts did not fit, but your tip on how to start GfxWrapper helped, thanks
took your martin-laclaustra_ESP32Lib-advanced and in 1bitmode everything works

pierreH23, 2021-01-21:
Hello,
Hope it helps : I managed to create new fonts with TTF2BMH (https://github.com/jdmorise/TTF2BMH) !
Now, my recipe for a font.ttf :

- adapt 'print_char' at the end of 'ttf2bmh.py' (# -> 1, . -> 0), comment print(char + ":") (line 204)
- run : python3 ttf2bmh.py -p -s 24 --square --ascii --progmem -f . > font.h where your font.tff stays
- insert first and last line : see in other fonts files to find teh correct syntax
- include font.h like others

Regards
Pierre

Martin-Laclaustra, 2021-01-25:
Thank you! We will test it.