bitluni/ESP32Lib #58: No Signal TV Out with Composite

osmaga2012, 2021-01-07:
Hi.
First of all, I want to thank the work done with this liberia, which I am testing.

I have made a simple sketch, a "Hello World" to get text using composite, but it always appears without signal on TV.

I put the code that I am using to see if you help me. I have connected pin 25 to outer wire and GND pin to inner.

`#include <ESP32Lib.h>

#include <Ressources/Font8x8.h>

CompositePAL8 graphics;

void setup()
{
    graphics.init(graphics.MODEPAL312P,graphics.GameWing);
    
    graphics.setFont(Font8x8);

    graphics.println("Hola mundo");
    
}

void loop()
{
  //graphics.println("Hola mundo");
}`

Thanks in advance!!!

Martin-Laclaustra, 2021-01-11:
Use "development" branch.
There are working examples there.

osmaga2012, 2021-01-12:
Thanks @Martin-Laclaustra hadn't seen the development branch ...

One more question. Is there a method where frames are drawn, for example a drawFrame() or something similar?

iProgramMC, 2022-03-23:
> Thanks @Martin-Laclaustra hadn't seen the development branch ...
> 
> One more question. Is there a method where frames are drawn, for example a drawFrame() or something similar?

`setFrameBufferCount(nBuffers)` allows you to use double buffering.  In your loop you call `vga.show()` as soon as you're done drawing