bitluni/ESP32Lib #59: #pragma one not working?

osmaga2012, 2021-01-14:
Hi. Thanks for the work done with this library.

I am doing a project, where I use this library with separate files. Let me explain with an example.

I've two files, 

Main.ino
`#include <ESP32Lib.h>

extern CompositeGrayDAC display;
extern void display_begin();

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}`

And other file, 

display.cpp,

`#include <ESP32Lib.h>
#include "Ressources/Font8x8.h"

CompositeGrayDAC display;

extern void display_begin(){

  display.init(CompMode::MODEPAL288P, 25, false);

  display.setFont(Font8x8);
  display.setCursor(50, 100);

  display.print("Test");
  
}`

Both files are in the same project directory ...

When I compile I get the following error message and I suspect it is from the headers...

`Arduino:1.8.13 (Windows 10), Tarjeta:"ESP32 Dev Module, Enabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

sketch\emu_nes3.ino.cpp.o: In function `VGA14BitI::interruptPixelLine(int, unsigned long*, unsigned long, void*)':

C:\Users\oscar\Documents\Arduino\libraries\bitluni_ESP32Lib\src/VGA/VGA14BitI.h:223: multiple definition of `VGA14BitI::interruptPixelLine(int, unsigned long*, unsigned long, void*)'

sketch\display.cpp.o:C:\Users\oscar\Documents\Arduino\libraries\bitluni_ESP32Lib\src/VGA/VGA14BitI.h:223: first defined here

sketch\emu_nes3.ino.cpp.o: In function `VGA14BitI::interruptPixelLine(int, unsigned long*, unsigned long, void*)':

C:\Users\oscar\Documents\Arduino\libraries\bitluni_ESP32Lib\src/VGA/VGA14BitI.h:226: multiple definition of `VGA14BitI::interrupt(void*)'

sketch\display.cpp.o:display.cpp:(.iram1+0x3c): first defined here

sketch\emu_nes3.ino.cpp.o: In function `VGA14BitI::interrupt(void*)':

C:\Users\oscar\Documents\Arduino\libraries\bitluni_ESP32Lib\src/VGA/VGA14BitI.h:215: multiple definition of `VGA6BitI::interruptPixelLine(int, unsigned long*, unsigned long, void*)'

sketch\display.cpp.o:display.cpp:(.iram1+0xb0): first defined here

sketch\emu_nes3.ino.cpp.o: In function `VGA6BitI::interruptPixelLine(int, unsigned long*, unsigned long, void*)':

C:\Users\oscar\Documents\Arduino\libraries\bitluni_ESP32Lib\src/VGA/VGA6BitI.h:229: multiple definition of `VGA6BitI::interrupt(void*)'

sketch\display.cpp.o:display.cpp:(.iram1+0x104): first defined here

sketch\emu_nes3.ino.cpp.o: In function `VGA6BitI::interrupt(void*)':

C:\Users\oscar\Documents\Arduino\libraries\bitluni_ESP32Lib\src/VGA/VGA6BitI.h:206: multiple definition of `VGA3BitI::interruptPixelLine(int, unsigned long*, unsigned long, void*)'

sketch\display.cpp.o:display.cpp:(.iram1+0x17c): first defined here

sketch\emu_nes3.ino.cpp.o: In function `VGA3BitI::interruptPixelLine(int, unsigned long*, unsigned long, void*)':

C:\Users\oscar\Documents\Arduino\libraries\bitluni_ESP32Lib\src/VGA/VGA3BitI.h:212: multiple definition of `VGA3BitI::interrupt(void*)'

sketch\display.cpp.o:C:\Users\oscar\Documents\Arduino\libraries\bitluni_ESP32Lib\src/Composite/CompositeGrayDAC.h:133: first defined here

sketch\emu_nes3.ino.cpp.o: In function `VGA3BitI::interrupt(void*)':

C:\Users\oscar\Documents\Arduino\libraries\bitluni_ESP32Lib\src/VGA/VGA3BitI.h:189: multiple definition of `VGA1BitI::interruptPixelLine(int, unsigned long*, unsigned long, void*)'

sketch\display.cpp.o:C:\Users\oscar\Documents\Arduino\libraries\bitluni_ESP32Lib\src/Composite/CompositeGrayDAC.h:336: first defined here

sketch\emu_nes3.ino.cpp.o: In function `VGA1BitI::interruptPixelLine(int, unsigned long*, unsigned long, void*)':

C:\Users\oscar\Documents\Arduino\libraries\bitluni_ESP32Lib\src/VGA/VGA1BitI.h:208: multiple definition of `VGA1BitI::interrupt(void*)'

sketch\display.cpp.o:c:\users\oscar\documents\arduino\libraries\bitluni_esp32lib\src\graphics/Graphics.h:807: first defined here

sketch\emu_nes3.ino.cpp.o: In function `VGA1BitI::interrupt(void*)':

C:\Users\oscar\Documents\Arduino\libraries\bitluni_ESP32Lib\src/VGA/VGA1BitI.h:189: multiple definition of `VGATextI::interruptPixelLine(int, unsigned long*, unsigned long, void*)'

sketch\display.cpp.o:c:\users\oscar\documents\arduino\libraries\bitluni_esp32lib\src\graphics/Graphics.h:111: first defined here

sketch\emu_nes3.ino.cpp.o: In function `VGATextI::interruptPixelLine(int, unsigned long*, unsigned long, void*)':

C:\Users\oscar\Documents\Arduino\libraries\bitluni_ESP32Lib\src/VGA/VGATextI.h:293: multiple definition of `VGATextI::interrupt(void*)'

sketch\display.cpp.o:c:\users\oscar\documents\arduino\libraries\bitluni_esp32lib\src\graphics/Graphics.h:937: first defined here

sketch\emu_nes3.ino.cpp.o: In function `VGATextI::interrupt(void*)':

C:\Users\oscar\Documents\Arduino\libraries\bitluni_ESP32Lib\src/VGA/VGATextI.h:209: multiple definition of `timerInterrupt(AudioOutput*)'

sketch\display.cpp.o:c:\users\oscar\documents\arduino\libraries\bitluni_esp32lib\src\graphics/Graphics.h:947: first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compilando para la tarjeta ESP32 Dev Module.
`

Martin-Laclaustra, 2021-01-14:
This is a design problem that will be corrected soon in the development process. There are some class function definitions within header files. Thus any "cpp" file, which in c++ is compiled separately from the rest ("ino" is another cpp block), can potentially duplicate that definition (read implementation). You have duplicated symbols and the linker fails. The wards (equivalent to that pragma) only work within a compilation block. For the moment, include this library only once in your project. Or create your "fragments" as header files (.h), and include them in the main "ino". That will avoid the need for the "extern".

osmaga2012, 2021-01-14:
Thanks @Martin-Laclaustra . I will wait for these modifications. If I can help with something I am totally available.

I do not understand very well this last thing that you comment. Do you have an example?

Thanks again.

Martin-Laclaustra, 2021-01-15:
> Thanks @Martin-Laclaustra . I will wait for these modifications. If I can help with something I am totally available.
Sometime in the future some help in systematic testing of modes will be welcome.
> I do not understand very well this last thing that you comment. Do you have an example?
Just rename display.cpp to display.h and include that file (with an include statement) on top of the ino file.