bitluni/ESP32Lib #61: Using 3d Engine with higher resolution

MagnusM1709, 2021-02-16:
Hello,
i want to use the VGA 3DEngine with another resolution than the default (200x150). 
No matter what resolution (except the standard resolution) I use, I always get a black image.

Any ideas?

(I use the Black Edition VGA board)

Martin-Laclaustra, 2021-02-17:
Check the serial output. You are probably running out of memory.

devfoFikiCar, 2021-09-07:
> Check the serial output. You are probably running out of memory.

What model then can support higher resolution, I have standard Do it ESP32 DEVKIT v1

iProgramMC, 2022-03-23:
I've seen some with 320 Kb and I've seen some with 520 Kb. Wikipedia says 320 Kb, other sites say 520 Kb. Did you try other resolutions and bitdepths, such as 320x240?