bitluni/ESP32Lib #62: Compile error while using lastest development branch

Lucy2003, 2021-02-27:
Hi,I'm trying to compile the example of this library but failed.

There is my Arduino console output : [Compile_error_report.zip](https://github.com/bitluni/ESP32Lib/files/6053651/Compile_error_report.zip)

Environment:
- ESP32 DevKit v1
- Arduino IDE 1.8.12
- esp_arduino core 1.0.4
- Windows 7 OS

Master branch is OK,but it doesn't support Composite output.

What I want to know more about the library is,
can I wire it as below? In that case,I can add a switch to allow myself change output mode between VGA and NTSCColor240P.

------------+      +--------+                         +---------+
 E      PinA0 | --- |   R2R    |   ------R-------- |  switch   | ---R--
 S      PinA1 | --- |              | -------G-------  |               | ---G--    VGA
 P      PinAx | --- |              | -------B-------- |               | ---B--
32   PinA13 | --- |              |                         +---------+
------------+     +---------+                          |     |     |
                                                                    +---+--+---------- Composite

Thank you.

Martin-Laclaustra, 2021-02-27:
In your compiler, several warnings are treated as errors.
However this is a good catch! These lines are ambiguous and it is worth to write them right.
I pushed code with the offending lines corrected. Could you please test and report back? (I would appreciate if you could test as many modes as you can)

With respect to your question... it seems difficult to share the ladder. I would recommend to use the DAC for composite and that way you can use both modes.

Lucy2003, 2021-02-28:
After your commit,almost all examples compile done but with some warnings.It is in
[Report.zip](https://github.com/bitluni/ESP32Lib/files/6056243/Report.zip)/Common_Warnings.txt
**But there are two example about VGA compile failed.**
Two LED examples is not tested.
I also copy my esp_arduino core's platform.txt to the zip.In this file it defined all the compiler args.It may help you test it on your PC.

Thanks for correcting ! :+1:

Martin-Laclaustra, 2021-03-01:
> **But there are two example about VGA compile failed.**

I solved the compiler errors (reorder errors) and pushed the updated code.
Thank you for your interest and help. Reviewing and fixing the warnings will require more time and though.

> I also copy my esp_arduino core's platform.txt to the zip.In this file it defined all the compiler args. It may help you test it on your PC.

Comparison of your platform.txt with mine did not show major differences.
The key setting to see the warnings is in the Arduino IDE: File > Preferences > (Tag) Settings > (drop-down list) Compiler warnings: "More"
It is interesting that with this option set to "None", the (previous unmodified) code compiled without errors.

> I know it's difficult to calculate the output voltage on R2R.

That is not the main point. You could re-use the 6-bits R2R for the VGA green channel and multiplex it for a composite signal (not simultaneously and with different code). Settings for the composite class must account for the 6-bit instead of the 8-bit ladder (I think it can be done, and it would probably work), and thus you lose signal depth resolution. If you want to try, it sounds like something to test. Go for it if you want and I could help but I am currently working in other parts of the library right now and I will not pursuit it on my own.

> ESP32 doesn't have enough resources to output double sound channel audio when outputing 480P VGA video.

I do not think you are right. Try using the ULP driver I adapted for ESP8266Audio:
https://github.com/bitluni/ULPSoundESP32/issues/7#issuecomment-785338253

It would be nice to know how your code evolves. Will you keep us updated, please?

Lucy2003, 2021-03-14:
I'm sorry for the late reply.I'm too busy to check my mailbox. Will you forgive me,please?

> I solved the compiler errors (reorder errors) and pushed the updated code.

I had tested them and it is OK! Thanks for the fix.

> The key setting to see the warnings is in the Arduino IDE: File > Preferences > (Tag) Settings > (drop-down list) Compiler warnings: "More"
It is interesting that with this option set to "None", the (previous unmodified) code compiled without errors.

I always set this to "All". That may be the reason I see these error.
Here are build options I copy from my Arduino IDE output (I didn't copy -I options because they are too long and included some paths)
```
-DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-missing-field-initializers -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=160000000L -DARDUINO=10812 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DESP32 -DCORE_DEBUG_LEVEL=0
```

>  Try using the ULP driver I adapted for ESP8266Audio

That's great! It's worth to use the library even if I need to reload my ULP program into slow rtc memory before my ESP32 going to deepsleep.

I will open a new issue when I need more help.Thanks again for your help!

closed 2021-03-14 by Lucy2003