bitluni/ESP32Lib #65: Show a bmp, png, jpg or sprite from SPIFFS

e-labInnovations, 2021-06-18:
I would like to show an image (BMP, png, jpg or sprite) from SPIFFS to VGA (MODE640x400 or any).
How is this possible?

Martin-Laclaustra, 2021-06-18:
This is an old goal. It would be great if you can help.
Try using the "examples/GFXWrapper", and the Adafruit_ImageReader library.
https://learn.adafruit.com/adafruit-gfx-graphics-library/loading-images
Please keep us posted of your progress.

lucadentella, 2021-06-20:
Hi, my library works with Adafruit GFX but I cannot test with the GFXWrapper. You opened an issue without further information... which error do you get? Where can I found your code?

Martin-Laclaustra, 2021-06-20:
@lucadentella
Hello Luca. I invited e-labInnovations to look into this topic and he found your library and wrote to you.
I read that you developed with ESP32. Thus, you could test this library without any hardware (just some jumper wires to a VGA cable or a composite TV). Clone the "development" branch (not the master) and have a look at the examples. In the src/vga and src/composite folders, header files have some simple schematics.
It would be really nice if your library and this one could work together.

lucadentella, 2021-06-21:
Hi Martin, unfortunately at the moment I'm on a job travel and I won't be able to work on this for at least a month: that's why I was asking @e-labInnovations if he made some tests with the two libraries or he was "only" asking for anyone to do the job for him...

e-labInnovations, 2021-06-21:
Hi @lucadentella 
I tried your library with this library. Here is my code
```
#include <ESP32Lib.h>
#include <GfxWrapper.h>
#include <SPIFFS.h>
#include <SPIFFS_ImageReader.h>

//VGA Device
VGA3Bit vga;
GfxWrapper<VGA3Bit> gfx(vga, 640, 400);
// Image reader
SPIFFS_ImageReader reader;

//pin configuration
#define redPin  12
#define  greenPin  14
#define  bluePin  27
#define  hsyncPin  26
#define  vsyncPin  25

//initial setup
void setup() {
  // initialize SPIFFS
  if(!SPIFFS.begin()) {
    Serial.println("SPIFFS initialisation failed!");
    while (1);
  }
  vga.init(vga.MODE640x400, redPin, greenPin, bluePin, hsyncPin, vsyncPin);

  // draw image
  reader.drawBMP("/minion.bmp", gfx, 0, 0);
}

void loop() {

}
```
Here is the compilation error
```
ESP32_VGA_SPIFFS_Bitmap:29:42: error: no matching function for call to 'SPIFFS_ImageReader::drawBMP(const char [12], GfxWrapper<VGA3Bit>&, int, int)'
   reader.drawBMP("/minion.bmp", gfx, 0, 0);
                                          ^

\SPIFFS_ImageReader_Library\src/SPIFFS_ImageReader.h:97:19: note: candidate: ImageReturnCode SPIFFS_ImageReader::drawBMP(char*, Adafruit_SPITFT&, int16_t, int16_t, boolean)
   ImageReturnCode drawBMP(char *filename, Adafruit_SPITFT &tft, int16_t x,
                   ^
```

lucadentella, 2021-06-22:
Hi

unfortunately my library requires an **Adafruit_SPITFT** object (the second parameter of the drawBMP method).

That is an intermediate object included in the Adafruit_GFX library:
http://adafruit.github.io/Adafruit-GFX-Library/html/class_adafruit___s_p_i_t_f_t.html

From what I understood, the GfxWrapper provided with this library is a wrapper for the "main" class (Adafruit_GFX) and therefore cannot be used directly with my library. You should write a "fake" SPITFT object or rewrite part of my library.