bitluni/ESP32Lib #71: Feature: support for videowall

wgroenewold, 2021-11-18:
Hi,

Following projects like https://github.com/info-beamer/multiscreen and https://github.com/reinzor/videowall I was wondering if you would be interested in making ESP32Lib compatible with a multiscreen videowall setup. That way we don't have to use RasPi's for every screen but we can do it with ESP32's and we can recycle old LCD's that are around for nothing into something interesting. Would love your thoughts on this.

iProgramMC, 2022-03-23:
Sorry to break it to you, but the esp32 is not powerful enough to handle large resolutions such as those.  You're better off using Pi Zeros

wgroenewold, 2022-04-26:
@iProgramMC ESP32 should be enough to drive a single screen at 800x600 I think. The preprocessing (cutting a part of the complete image and sending the apropriate part to a single screen) could be done with a Rpi or another random (more powerful) computer.