bitluni/ESP32Lib #72: [Question] VGA to HDMI converter not working

jsbriantes, 2021-12-14:
I have this [VGA2HDMI ](https://es.aliexpress.com/item/32848408479.html?spm=a2g0o.productlist.0.0.56d86c15fhB1Tu&algo_pvid=7a56b9cc-be10-4cf9-90cd-3d9fc4c8c316&aem_p4p_detail=20211214010547888649721680017252441&algo_exp_id=7a56b9cc-be10-4cf9-90cd-3d9fc4c8c316-1&pdp_ext_f=%7B%22sku_id%22%3A%2267226878657%22%7D) converter that I use to conect a 5" hdmi screen to a VGA source. It works good with my old laptop, and I can see the Windows 7 screen on the small display using this VGA2HDMI converter ( 800x600).
I would like to use this small screen with ESP32 and this library. I have used basic demo example with an ESP32 and a TFT VGA screen with a direct wire connection (3bit mode), and It works good too. Also, advanced examples works good on the 17" screen.
Now, I would like use the vga2hdmi converter, but It doesn't work. I have tried adding +5VDC to pin 9, but without success. I don't know the reason, but I have checked that ESP32 and VGA plug works good, and converter works good too.
I was wondering if, perhaps, the ESP32 can't supply enough current to RGB pins or something else. Perhaps I would need to use I2C for the other pins.
I know It is a very general question, but perhaps, someone has tried before, what I want.

jsbriantes, 2021-12-14:
Researching more on this topic, perhaps the problem could be HSYNC and VSYNC that should be TTL signals and my VGA2HDMI converter doesn't work with 3.3V signals, while my old 17" vga screen accept this logic levels. I will try with a logic level converter for 3.3V to 5V.

jsbriantes, 2021-12-14:
Failed test. It doesn't work. I don't know what else could I try.

dl9sec, 2021-12-15:
Could it be possible, that the VGA2HDMI doesn't support the video mode you provide with the ESP32? Which VGA mode do you want to use (resolution@frequency)?

Regards, Thorsten

jsbriantes, 2021-12-15:
These are the supported resolutions of VGA2HDMI device:
640x480@60/70/72/85Hz
800x600@60/70/72/85Hz
1024x768@60/70/72/85Hz
1152x864@75Hz
1280x720@60Hz
...
up to 1920x1080@60Hz

This is the display:  5" hdmi 800x480 xpt2046  for Raspberry Pi 3/2 Model B/B+ & Banana Pi
It's a very common display you could find it in many commerce websites. For example: https://es.aliexpress.com/item/1005001587506476.html?ds_e_product_id=es1005001587506476

I have checked it with my old laptop (sony vgn21 - Windows 7) with a screen resolution of 800x600 (60Hz), and It works good. Altough my laptop has a HDMI output, I have used VGA output and VGA2HDMI device converter to test it, with success.

dl9sec, 2021-12-15:
Ok, and which resolution@frequency you want to output at your ESP32?
Have you tried another, maybe shorter VGA cable?

jsbriantes, 2021-12-15:
I want to use 800x600@60Hz. I'm using a very short cable, about 20cm, that works good with 800x600 old 17" TFT Screen.
I don't know if the VGA signal is generated at this rate (60Hz) or less. There is no info about it in the main page of the project (only supported resolutions). Perhaps I could find it reading inside library files. But I think It should meet VGA standard and this should be the rate.

Thinking about the plug, GNDs for Red, Green and Blue are not connected (pin 6, 7 and 8). Should I connect it to common GND?. Perhaps it could be the problem, and the VGA2HDMI converter needs this lines connected to GND insted of N/C. I don't know. I will try it too.

dl9sec, 2021-12-15:
Maybe there is the possibility at your 17" screen to check the VGA signal parameters (horizontal/vertical frequencies, ...). Possibly the VGA2HDMI isn't as tolerant as your monitor...

jsbriantes, 2021-12-15:
The screen is ACER V173 Bb. I thought it could be H/V SYNC at 3.3V. HSYNC and VSYNC signals are a train of squared pulses of +5V and I was using 3.3V. Then I added a 3.3V to 5V level converter, but without success.
Perhaps, as you said, H/V frequencies are not accurate enough from ESP32 and the VGA2HDMI converter can't sycronize it. I have no way to measure it (I don't have an oscilloscope).

fjmazur, 2021-12-28:
I have same problem here.. I wait for a hdmi solution too.. Thanks

jsbriantes, 2021-12-28:
I'm sorry but I haven't found the solution. I don't have an oscilloscope and then I can't take some measurements in order to know where is the problem. At last, I have moved to the new Raspberry Pi Zero 2W that has a HDMI builtin output and It's cheap. I think ESP32 doesn't have enough power (Mhz) to drive HDMI directly. You will need some additional hardware and the cost it has is more than the Raspberry Pi Zero 2W.

Gabryxx, 2022-05-07:
I buy this adapter and work:
https://www.amazon.it/gp/product/B09S6SNB7L/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1
I have made this connections (also pin 6-7-8 with ground of VGA connector):
[image]
And this is resolution that i use:

VGA3Bit vga;
GfxWrapper<VGA3Bit> gfx(vga, 400, 300);
void setup() {
  vga.setFrameBufferCount(2);// DOUBLE BUFFER
  vga.init(vga.MODE400x300, 27, 14, 12, 32, 33);

But problem of this library is that cannot use SPIFFS, SD, Wifi (lost connection), so, if you need use em, need consider another ESP32 Dev Kit V1 wired with UART2:
so one ESP32 dedicated to the management of the VGA and the main program and the other to the management of the SPIFF or SD etc ... and in this way the variables are exchanged via UART 2

yo2ldk, 2022-08-26:
I also need to convert from VGA  to HDMI, if you find any solution..