bitluni/ESP32Lib #75: Issue while trying to compile for ESP32-C3 Dev board

Technerder, 2022-05-24:
I was trying to run the basic `VGAHelloWorld` sketch on my `esp32-c3 devkitc-02` board and ran into the following issue. How can I solve this problem?

```output
In file included from C:\Users\User\Documents\Arduino\libraries\bitluni_ESP32Lib\src/ESP32Lib.h:21,
                 from C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_400707\VGAHelloWorld.ino:6:
C:\Users\User\Documents\Arduino\libraries\bitluni_ESP32Lib\src/Audio/AudioOutput.h: In function 'void timerInterrupt(AudioOutput*)':
C:\Users\User\Documents\Arduino\libraries\bitluni_ESP32Lib\src/Audio/AudioOutput.h:51:42: error: no match for 'operator=' (operand types are 'volatile timg_txupdate_reg_t' and 'int')
       TIMERG0.hw_timer[TIMER_0].update = 1;
                                          ^
In file included from C:\Users\User\Documents\Arduino\libraries\bitluni_ESP32Lib\src/Audio/AudioOutput.h:13,
                 from C:\Users\User\Documents\Arduino\libraries\bitluni_ESP32Lib\src/ESP32Lib.h:21,
                 from C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_400707\VGAHelloWorld.ino:6:
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/timer_group_struct.h:97:3: note: candidate: 'timg_txupdate_reg_t& timg_txupdate_reg_t::operator=(const timg_txupdate_reg_t&)'
 } timg_txupdate_reg_t;
   ^~~~~~~~~~~~~~~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/timer_group_struct.h:97:3: note:   no known conversion for argument 1 from 'int' to 'const timg_txupdate_reg_t&'
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/timer_group_struct.h:97:3: note: candidate: 'timg_txupdate_reg_t& timg_txupdate_reg_t::operator=(timg_txupdate_reg_t&&)'
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/timer_group_struct.h:97:3: note:   no known conversion for argument 1 from 'int' to 'timg_txupdate_reg_t&&'
In file included from C:\Users\User\Documents\Arduino\libraries\bitluni_ESP32Lib\src/ESP32Lib.h:21,
                 from C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_400707\VGAHelloWorld.ino:6:
C:\Users\User\Documents\Arduino\libraries\bitluni_ESP32Lib\src/Audio/AudioOutput.h:52:30: error: 'volatile union timg_int_clr_timers_reg_t' has no member named 't0'
       TIMERG0.int_clr_timers.t0 = 1;
                              ^~
C:\Users\User\Documents\Arduino\libraries\bitluni_ESP32Lib\src/Audio/AudioOutput.h:53:40: error: 'volatile union timg_txconfig_reg_t' has no member named 'alarm_en'
       TIMERG0.hw_timer[TIMER_0].config.alarm_en = 1;
                                        ^~~~~~~~
In file included from C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/dport_access.h:22,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/hal/esp32c3/include/hal/cpu_ll.h:11,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/hal/include/hal/cpu_hal.h:16,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/esp_hw_support/include/esp_cpu.h:14,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/esp_hw_support/include/soc/cpu.h:14,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/esp_hw_support/include/soc/spinlock.h:11,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/freertos/port/riscv/include/freertos/portmacro.h:38,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/freertos/include/freertos/portable.h:51,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/freertos/include/freertos/FreeRTOS.h:63,
                 from C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1\cores\esp32/Arduino.h:33,
                 from sketch\VGAHelloWorld.ino.cpp:1:
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/soc.h:75:50: error: 'DR_REG_I2S_BASE' was not declared in this scope
 #define REG_I2S_BASE(i)                         (DR_REG_I2S_BASE + (i) * 0x1E000)
                                                  ^~~~~~~~~~~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/soc.h:81:34: note: in definition of macro 'ETS_UNCACHED_ADDR'
 #define ETS_UNCACHED_ADDR(addr) (addr)
                                  ^~~~
C:\Users\User\Documents\Arduino\libraries\bitluni_ESP32Lib\src/Audio/AudioOutput.h:55:7: note: in expansion of macro 'WRITE_PERI_REG'
       WRITE_PERI_REG(I2S_CONF_SIGLE_DATA_REG(0), audioOutput->audioSystem->nextSample() << 24);
       ^~~~~~~~~~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/i2s_reg.h:1020:46: note: in expansion of macro 'REG_I2S_BASE'
 #define I2S_CONF_SIGLE_DATA_REG(i)          (REG_I2S_BASE(i) + 0x0068)
                                              ^~~~~~~~~~~~
C:\Users\User\Documents\Arduino\libraries\bitluni_ESP32Lib\src/Audio/AudioOutput.h:55:22: note: in expansion of macro 'I2S_CONF_SIGLE_DATA_REG'
       WRITE_PERI_REG(I2S_CONF_SIGLE_DATA_REG(0), audioOutput->audioSystem->nextSample() << 24);
                      ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/soc.h:75:50: note: suggested alternative: 'DR_REG_I2S0_BASE'
 #define REG_I2S_BASE(i)                         (DR_REG_I2S_BASE + (i) * 0x1E000)
                                                  ^~~~~~~~~~~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/soc.h:81:34: note: in definition of macro 'ETS_UNCACHED_ADDR'
 #define ETS_UNCACHED_ADDR(addr) (addr)
                                  ^~~~
C:\Users\User\Documents\Arduino\libraries\bitluni_ESP32Lib\src/Audio/AudioOutput.h:55:7: note: in expansion of macro 'WRITE_PERI_REG'
       WRITE_PERI_REG(I2S_CONF_SIGLE_DATA_REG(0), audioOutput->audioSystem->nextSample() << 24);
       ^~~~~~~~~~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/i2s_reg.h:1020:46: note: in expansion of macro 'REG_I2S_BASE'
 #define I2S_CONF_SIGLE_DATA_REG(i)          (REG_I2S_BASE(i) + 0x0068)
                                              ^~~~~~~~~~~~
C:\Users\User\Documents\Arduino\libraries\bitluni_ESP32Lib\src/Audio/AudioOutput.h:55:22: note: in expansion of macro 'I2S_CONF_SIGLE_DATA_REG'
       WRITE_PERI_REG(I2S_CONF_SIGLE_DATA_REG(0), audioOutput->audioSystem->nextSample() << 24);
                      ^~~~~~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board ESP32C3 Dev Module.
```

bitluni, 2022-05-24:
I would love to support c3 at some point, unfortunately the esp32lib is
optimized to the original esp32
If you are proficient in coding you could remove the audio part that's
causing errors and find the right registers. I'm not sure if the c3 has two
I2S channels. There are certainly some differences.
my new approach would be to try using espressif api most of the time since
they are covering all of the flavors of ESPs there are now

JoshuaColell, 2022-08-11:
> I'm not sure if the c3 has two I2S channels

The esp32-c3 only has one I2S peripheral, while the esp32 has two.
Maybe [this guide](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32c3/api-reference/peripherals/i2s.html) could help