bitluni/ESP32Lib #79: [Question]How to make 8bit(256color) mode?

riraosan, 2022-12-01:
Hi. @bitluni 

I would like to use your results to create an 8BIT(256colors) mode VGA class.
Rough question, do I need to create a VGA class for 8BIT mode?
Or can I use it as 8BIT mode by omitting the pins on the 554 in 14 bit mode?

riraosan, 2022-12-01:
For example, can I use 14bit mode VGA as 8bit mode using the ports R2, R3, R4, G2, G3, G4, B2 ,B3, hsync, vsync?

Martin-Laclaustra, 2022-12-02:
Have you tried [6BIT mode](https://github.com/bitluni/ESP32Lib/tree/master/examples/6BitMode)?
What do you aim to do?
- In terms of memory savings, you will not get any with 8bit, as you will be using 10 bits and wasting the rest of the 16 bits that support 14bit mode.
- If your original image data is in 8bit, you can decimate it to 6bit easily.
- If your original image data is in 8bit, and you increase color depth to 14bit naively (bit-shifting), when you render it in 14bit hardware you would lose the top brightness as you will have the last low bits off... if you need more accuracy you will need to write some conversion algorithm. If you render it in the hardware that you propose you may be ok, but you will be redefining the meaning of the colors and that may bring trouble if you use other conversion functions of the library (in terms of color inaccuracy).
- If you use 14bit images in your proposed incomplete 8bit output hardware you probably will be ok. (however, just buy yourself some more resistors and implement the complete 14bit output!)

Martin-Laclaustra, 2022-12-02:
Check the development branch for a bunch of other options.

riraosan, 2022-12-05:
Hi @Martin-Laclaustra 
> Have you tried [6BIT mode](https://github.com/bitluni/ESP32Lib/tree/master/examples/6BitMode)?

No, I have not tried 6bit mode.
I tried 14bit mode.
Of course, when I used 14bit mode on the circuit for the 332, the white background was not white and the lines drew a [pattern](https://twitter.com/riraosan_0901/status/1599341113382694912?s=20&t=f2utKpg8dA61strHYkhGgA).
I understand a little more what you are saying. Thank you very much.

I have confirmed that the RGBA() API converts 16 bits of pixel color to 14 bits.
I assume that the process of rounding to 6 bits is the same. After this I would like to confirm.

I have three main objectives. 
1. to save memory (this seems impossible in 8bit mode)
1. to connect and use the VGA library and lovyanGFX(very fast GFX Library for M5Stack. [8bit Composite and lovyanGFX sample movie](https://www.instagram.com/p/CbXvBUovzNE/?utm_source=ig_web_copy_link))
1. enjoy (JOKE)

1. doesn't seem possible. 2. could do it.

I would like to replace your GFX with lovyanGFX using 6bit mode after this.

I appreciate your contribution. Thank you very much.

@riraosan