bitluni/ESP32Lib #81: ESP32S3 Support

BrainBoardz, 2022-12-15:
I've posted this in the FabGL forum as well. , It would be amazing if we could get ESP32S3 support added to to ESP32Lib. The S3 has two I2S channels and a lot of free pins. It lacks a hardware DAC, put perhaps an I2S for Audio and an I2S for VGA, so using both i2s channels. A VGA board based around an S3 would be really cool.

BrainBoardz, 2022-12-21:
I did some ad hoc testing (not my area of experience!) with the VGA library and after removing the audio importing there wasn't a huge number of errors and these were al in the i2s.cpp file. Most of the errors appear to be definition and naming issues. I'm guessing that the ESP32 and ESP32S3 have support libraries with different naming. So, I think it maybe possible to adapt the VGA library to ESP32S3.

zeezooX, 2023-04-18:
Have you managed to make it work?

BrainBoardz, 2023-04-19:
No. I've been too busy with other project to take it further than just verifying that most of the issues are related to audio and perhaps some definition differences with the regular ESP32.  Would be great if VGA worked though!

Mer08, 2023-11-23:
Bitluni made a library for this

odelot, 2024-06-29:
@joveaaron do you mean https://github.com/bitluni/ESP32-S3-VGA?

BrainBoardz, 2024-07-04:
I believe there is some work going on to get the ESP32S3 to work with FabGL.  

The roadblock afaik is/was the lack of DACs on the S3, as the regular ESP32 used it's DAC for VGA timing and audio output. Bitfixer used a different approach than Bitluni to generate the sync for the VGA signal.  That code is clearly more efficient than Bitluni's method, but it relies on the ESP-IDF and there is no Arduino library support.  Also, it hasn't been updated in quite some time. 

That said, there is clearly a potential to get VGA+Audio working. Native support  for the S3 in FabGL would not only standardize the  VGA approach, pinouts etc. , but it would open up the S3 to the entire FabGL ecosystem potentially.  Basically, FabFGL on an ESP32S3 could be amazing!  I'm using the S3 with the octal PSRAM (N8/R8) on my prototype VGA board. I designed an I2S DAC specifically for this board (works great BTW!), but the VGA board project is on hold as software support isn't quite there. I did verify VGA and I2S audio works using Bitluni's code and some I2S code, but this only worked at 320X240 or thereabouts. I used a cheap VGA+Audio to HDMI converter. Looked and sounded fantastic!  

I see that someone posted recently about getting FabGL working on YouTube:  

https://www.youtube.com/watch?v=_lVKmdsJ320

As a side note. I'd love to see support in FabGL for Bluetooth joysticks on the S3. For gaming scenarios this would be very handy. I was able to get a Terios T3 working  on the S3 using BluePad32 thanks to a lot of help from the author. It was a pain to find a working controller as the S3 does not support Bluetooth Classic. Only the T3 worked for me.  BluePad32 is a big library.  

PS2 keyboard/mouse or USB support for the S3 in FabGL would also be great, of course, but support for a Bluetooth controller would make for a really nice gaming package and potentially allow for simpler board designs.