bitluni/ESP32Lib #82: ESP32-WROOM-32U-N4 Compatibility

JanoschABR, 2023-04-07:
I am thinking of making a board with VGA video output and want to use an ESP32-WROOM-32U-N4 as a sort of GPU for it. Would this module be suitable for this purpose or do I need to use another?

My goal is to output 800x600 video and being able to have another chip interface with it to change pixels and/or draw graphics. The module has 520 KB of RAM which should be more than enough to hold an 800x600 framebuffer...

I haven't quite understood yet how the VGA signals are actually generated in the ESP, so I can't really fathom what capabilities a module needs to have in order to be compatible (be it with a few changes to the code).

Any help/info is appreciated.

Tairex777, 2023-08-19:
The issue with the RAM is that there's really only 160KB available in this case.

"There is 520 KB of available SRAM (320 KB of DRAM and 200 KB of IRAM) on the ESP32. However, due to a technical limitation, the maximum statically allocated DRAM usage is 160 KB. The remaining 160 KB (for a total of 320 KB of DRAM) can only be allocated at runtime as heap."
(Memory Types - ESP32 - — ESP-IDF Programming Guide Latest Documentation, n.d.)

I looked at the code and it uses malloc() to allocate framebuffers. This places the framebuffers in heap memory, which is limited to 160KB.

I personally have used 320x200 resolution with double buffering, and 400x300 with single buffering. 400x300 with double buffering didn't work for me, and I'm not sure about higher resolutions. I've also had issues with low frame rates, but this could just be me.

Sources: 
Memory Types - ESP32 - — ESP-IDF Programming Guide latest documentation. (n.d.). Docs.espressif.com. Retrieved August 19, 2023, from https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/memory-types.html

JanoschABR, 2023-08-20:
Wow, completely forgot I had commented this.
I've since then moved over to the ESP32-S3 and have gotten everything to work.

closed 2023-08-20 by JanoschABR