bitluni/ESP32Lib #94: PLATFORMIO COMPILE ERRORS

Inktortoise, 2024-10-19:
I use PlatformIO and I got this errors when tried to compile:

```
.pio/libdeps/esp32dev/bitluni ESP32Lib/src/I2S/I2S_ESP32.cpp:379:27: error: too few arguments to function 'void rtc_clk_apll_enable(bool, uint32_t, uint32_t, uint32_t, uint32_t)'

.pio/libdeps/esp32dev/bitluni ESP32Lib/src/I2S/I2S_ESP32.cpp:380:3: error: 'rtc_clk_apll_coeff_set' was not declared in this scope

.pio/libdeps/esp32dev/bitluni ESP32Lib/src/I2S/I2S_ESP32.cpp:461:26: error: too few arguments to function 'void rtc_clk_apll_enable(bool, uint32_t, uint32_t, uint32_t, uint32_t)'

.pio/libdeps/esp32dev/bitluni ESP32Lib/src/I2S/I2S_ESP32.cpp:462:2: error: 'rtc_clk_apll_coeff_set' was not declared in this scope

*** [.pio\build\esp32dev\lib638\bitluni ESP32Lib\I2S\I2S_ESP32.cpp.o] Error 1
```

I'm new for this and don't know how to say my problem correctly, but my platformio.ini is this:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
    bitluni/bitluni ESP32Lib@^0.4.0
```

I tried using:

```
lib_deps =
    bitluni/bitluni ESP32Lib@^0.3.4
```

And it compiles finally, but when I tried to use Hello World example, it says:

`src/main.cpp:6:10: fatal error: ESP32Video.h: No such file or directory`

I don't know what to do...

Inktortoise, 2024-11-17:
I moved to Arduino IDE and It works now.

closed 2024-11-17 by Inktortoise

YonLiud, 2024-11-23:
It's not really a solution...
suggesting to re-open as it still occurs in newer releases

Jimsss1, 2025-01-12:
Because you are using version 0.3.4 of the library and the example you are using is version 0.4.0, this header file is not in 0.4.0